opendata-stuttgart/meta #130: sds011 lüfter springt an, aber keine Daten

andreas83, 2020-01-02:
Hallo und danke für das gelungene Projekt!

Ich hab mein Node 5635678 registriert, der DHT22 wird auch erkannt und die Daten werden ordentlich übermittelt. 
Der sds011 springt zwar an, lüfter läuft alle x min, aber es werden keine Daten erfasst. 
Meine Konfig:

`
Senden an Luftdaten.info: 1
Senden an Madavi: 1
Lese DHT: 1
Lese HTU21D: 0
Lese SDS: 1
Lese PMS(1,3,5,6,7)003: 0
Lese HPM: 0
Lese SPS30: 0
Lese PPD: 0
Lese BMP/E280: 0
Lese DS18B20: 0
Lese DNMS: 0
Korrekturwert in dB(A): 1
Lese GPS: 0
Auto Update: 1
Lade Beta Versionen: 0
OLED SSD1306: 0
OLED SH1106: 0
OLED Display um 180° drehen: 0
LCD 1602 (I2C: 0x27): 0
LCD 1602 (I2C: 0x3F): 0
LCD 2004 (I2C: 0x27): 0
Zeige WiFi Info: 1
Zeige Geräteinfo: 1
Debug Level: 5
Messintervall: 145000
Senden an Feinstaub-App: 0
Senden an aircms.online: 0
Senden an CSV: 0
Senden an Feinstaub-App: 0
Senden an opensensemap: 0
senseBox-ID

An eigene API senden: 1
Server: codejungle.org
Pfad: /track_bots.php
Port: 443
Benutzer:
Passwort:

InfluxDB: 0
Server: influx.server
Pfad: /write?db=luftdaten
Port: 8086
Benutzer:
Passwort:
Measurement: feinstaub
SSL: 0`

iceweasel1, 2020-01-02:
Welche Firmware läuft den aktuell bei dir?

andreas83, 2020-01-02:
Firmware: NRZ-2019-125-B1

Terminal
00:55:47.514 -> .......
00:55:47.514 -> 
00:55:47.514 -> WiFi connected
00:55:47.514 -> IP addres....
00:55:49.039 -> NTP time  ... not found
00:55:49.039 -> twoStageAutoUpdate
00:55:49.072 -> End reading SDS011 version date
00:55:49.669 -> 0 - 170 - aa - 170 .
00:55:49.669 -> 1 - 197 - c5 - 197 .
00:55:49.702 -> 2 - 6 - 6 - 6 .
00:55:49.702 -> 0 - 1 - 1 - 1 .
00:55:49.735 -> 0 - 1 - 1 - 1 .
00:55:49.735 -> 0 - 0 - 0 - 0 .
00:55:49.768 -> 0 - 197 - c5 - 197 .
00:55:49.801 -> 0 - 236 - ec - 236 .
00:55:49.801 -> 0 - 185 - b9 - 185 .
00:55:49.834 -> 0 - 171 - ab - 171 .
00:55:49.867 -> 0 - 170 - aa - 170 .
00:55:49.867 -> 1 - 197 - c5 - 197 .
00:55:49.901 -> 2 - 7 - 7 - 7 .
00:55:49.934 -> 3 - 18 - 12 - 18 .
00:55:49.934 -> 4 - 11 - b - 11 .
00:55:49.967 -> 5 - 16 - 10 - 16 .
00:55:50.000 -> 6 - 197 - c5 - 197 .
00:55:50.000 -> 7 - 236 - ec - 236 .
00:55:50.033 -> 8 - 229 - e5 - 229 .
00:55:50.066 -> Checksum is: 229
00:55:50.066 -> Checksum should: 229
00:55:50.100 -> 9 - 171 - ab - 171 .
00:55:50.133 -> SDS version date : 18-11-16
00:55:50.133 -> SDS device ID: c5ec
00:55:50.166 -> End reading SDS011 version date
00:55:50.398 -> Success downloading: /airrohr/update/latest_de.bin.md5
00:55:51.392 -> NewFW md5: e2982fe9300036ead4bfa1c58a92de92
00:55:51.525 -> Sketch md5    : e2982fe9300036ead4bfa1c58a92de92
00:55:51.558 -> No newer version available.
00:55:51.591 -> Starting Webserver... 192.168.0.234
00:55:51.624 -> 
00:55:51.624 -> ChipId: 5635678
00:55:51.707 -> Read SDS...
00:55:51.790 -> Stopping SDS011...
00:55:51.790 -> Read DHT...
00:55:51.823 -> Send to :
00:55:51.823 -> luftdaten.info
00:55:51.823 -> Madavi.de
00:55:51.856 -> custom API
00:55:51.856 -> ----
00:55:51.856 -> Auto-Update active...
00:55:52.851 -> Start reading SDS011
00:55:52.851 -> End reading SDS011
00:55:53.878 -> Start reading SDS011
00:55:53.878 -> End reading SDS011
00:55:54.872 -> Start reading SDS011
00:55:54.872 -> End reading SDS011
00:55:55.867 -> Start reading SDS011
00:55:55.867 -> End reading SDS011
00:55:56.861 -> Start reading SDS011
00:55:56.861 -> End reading SDS011
00:55:57.855 -> Start reading SDS011
00:55:57.855 -> End reading SDS011
00:55:58.882 -> Start reading SDS011
00:55:58.882 -> End reading SDS011
00:55:59.875 -> Start reading SDS011
00:55:59.875 -> End reading SDS011
00:56:00.869 -> Start reading SDS011
00:56:00.869 -> End reading SDS011
00:56:01.863 -> Start reading SDS011
00:56:01.863 -> End reading SDS011
00:56:02.857 -> Start reading SDS011
00:56:02.857 -> End reading SDS011
00:56:03.884 -> Start reading SDS011

andreas83, 2020-01-03:
Hab die Firmware auf Beta (NRZ-2019-128-B6)  gewechselt, nun werden auch werte angezeigt!

dirkmueller, 2020-01-08:
Es gibt nun einen NRZ-2020-XXX build (sowohl auf stable als auch auf beta). dieser sollte auch gehen. wenn auf diese version per OTA geupdated wurde und keine neuen probleme auftreten bitte das issue schliessen. Danke im Voraus!

iceweasel1, 2020-01-09:
Ähm ja. Problem besteht anscheinend noch.
[image]

Ist das Absicht, dass bei der 2020-Firmware die Messzeit und die wie vielte Messung es bis dato war, rausgemacht wurde in der Übersicht?

dirkmueller, 2020-01-09:
@iceweasel1 diese Angaben sind nun unter dem eigenen Menüpunkt 'Gerätestatus' zu finden.

Zu dem sds011 Problem.. https://github.com/opendata-stuttgart/sensors-software/issues/586 wartet noch auf Antwort..

iceweasel1, 2020-01-09:
@dirkmueller Ahh gefunden. So sieht es gerade bei mir aus. Die Uhrzeit im Status läuft als UTC?
[image]

586 kommt später.

dirkmueller, 2020-01-09:
@iceweasel1 jep, das Gerät läuft immer in UTC.

MarioA23, 2023-05-01:
Hallo zusammen!

Ich wollte auch mal ein wenig basteln und habe nun ebenfalls das Problem, dass der SDS011 bei mir keine Werte liefert. Sowohl mit der Firmware "latest.bin" als auch mit der "Beta-Version" (NRZ-2021-134-B4/DE (Aug 22 2021)) passiert nichts. Habe auch einen zweiten Sensor ausprobiert, selbes Problem (über USB am PC zeigen beide Werte an!). Kabel sind durchgemessen, alle Lötstellen nachgelötet, daran liegt's also auch nicht.

Hat jemand evtl. eine Idee??? Ich verzweifele!!!

Beste Grüße,
Mario

[image]
[image]

pjgueno, 2023-05-01:
Hi,
Can I see the config page.
Is RX->TX and TX->RX ?
The SDS is very sensitive for false Connections. Was it well plug on the first attempt ?

MarioA23, 2023-05-01:
Hello Pierre-Jean,
what do you want to see exactly? The wiring should be correct, TX goes to D1 and RX goes to D2.