opendata-stuttgart/meta #131: sds011 nimmt keine Feinstaubdaten mehr, Temp/Hum aber doch

zgadgeter, 2020-01-05:
Hallo,
mein Sensor hat jetzt ueber einem Jahr gut funktioniert, hat alle daten gesendet. Aber seid ca 1 Monat nimmt er keine Feinstaubdaten mehr auf. Temp/hum schon.
Geaendert habe ich nichts an dem Sensor. 
Ich habe schon die die Beta Firmware probiert, aendert aber nichts. (NRZ-2019-128-B7).
Hier ein Status von dem Device:
Firmware version	NRZ-2019-128-B7
SDK:2.2.2-dev(38a443e)
Core:2.6.2=20602000
lwIP:STABLE-2_1_2_RELEASE
glue:1.2-16-ge23a07e
BearSSL:89454af 
Free Memory	27544 
Heap Fragmentation	9 %
Last OTA	15 min, 15s 
NTP Sync	1 
Time	Sun Jan 5 16:19:24 2020  
Uptime	15 min, 17s 
Reset Reason	Software/System restart 
 
SDS011	 
 
only errors
WiFi	0 
OTA Return	200 
Data Send	0 
SDS011	6 
 
Number of measurements	6 
Time spent uploading	687 ms

iceweasel1, 2020-01-05:
Hallo @zgadgeter,

nehme dir mal eine viertel Stunde Zeit und lese schaue mal in die drei Threads. #551 #552 #586. Die sind alle drei von mir. Ich hatte am 1. November den Ersten eröffnet, da bei mir das selbe Spiel mit dem Sensor SDS011 ist. Das Problem ist hier, wie du in den drei Threads sehen wirst, bereits bekannt. Ich hatte von der normalen Firmware auf die Beta umgeschalten. In der Zwischenzeit kamen einige Betas dann nach und nach raus. 

Bisher habe ich das Problem weiterhin. Nur nicht mehr so stark, wie ich sonst in den drei Threads berichtet habe. Das Problem habe ich trotzdem weiterhin.

Grüße
René

iceweasel1, 2020-01-06:
Bei mir seit etwa gestern (Sonntag) 05.03 Uhr werden wieder keine PM-Werte (https://www.madavi.de/sensor/graph.php?sensor=esp8266-331968-sds011) mehr übertragen. Da wird heute dann eine automatisierte E-Mail kommen, dass der Sensor seit ü24h keine Daten mehr übertragen hat.

WLAN, BME etc. läuft und zeichnet weiterhin problemlos auf. Bist also nicht alleine. ;-)

ricki-z, 2020-01-06:
Hallo,
wurde denn schon mal die Verkabelung am SDS011 kontrolliert. Bei manchen Geräten löst sich nach einiger Zeit einer der Stecker durch die Vibrationen des Lüfters.

zgadgeter, 2020-01-06:
> Hallo,
> wurde denn schon mal die Verkabelung am SDS011 kontrolliert. Bei manchen Geräten löst sich nach einiger Zeit einer der Stecker durch die Vibrationen des Lüfters.

Ok, danke. Das muss ich am WE machen da ich unter der Woche nicht zuhause bin. Werde es probieren. danke

freemann, 2020-01-06:
I'm also having problems with my SDS011 sensor on the most recent B3 firmware.
But I can't read the language you use above and so i'm not able to help or solve my issue.

Could you please write in English?

ricki-z, 2020-01-07:
Short version in english: In some cases a wire may have slipped off on the SDS011. This is caused by the vibrations of the fan and depends on the quality of the dupont cables.