opendata-stuttgart/meta #132: Statistics on Grenoble sensors (old PM values)

MLaure19, 2020-01-17:
Hi,

I am working on a research project and trying to get monthly historical PM values for Grenoble city (Sensor 37467 as an example).

Do you know if this is possible to get those values ?

Many thanks,

Marie-Laure

ricki-z, 2020-01-18:
Please have a look at http://archive.sensor.community/ . There is a folder with zipped CSV files per month. They are huge but should save you some time. You can the needed data by filtering by location.

MLaure19, 2020-01-20:
This is exactly what I needed ! Do you know how to credit data if we are using them for a scientific publication ? Many thanks !

ricki-z, 2020-01-20:
You can mention our project as "Sensor.Community (formerly known as Luftdaten.info)" plus the URL https://sensor.community/ .