opendata-stuttgart/meta #134: 24h data missing

mczapala, 2020-04-04:
Querying for the 24h data (https://data.sensor.community/static/v2/data.24h.json) returns empty json file.

stevecrawshaw, 2020-04-05:
As also does the hourly data: https://data.sensor.community/static/v2/data.1h.json which I am using for a real time feed..

ricki-z, 2020-04-05:
the real time data is in the data.json feed.
the data.1h.json contains aggregated values (average over 60 minutes).

At the moment there is a very high load in all data centers worldwide (including those of Amazon and Google !!! just in case why we aren't switching to one of them). Thats why we needed to reduce the load on our servers.
As soon as the data center is upgrading the hardware we will enable these feeds again.

mczapala, 2020-04-05:
Any ETA when that might happen (days, weeks, months)?

ricki-z, 2020-04-05:
More than some days for the 24h values (but hopefully only a few weeks).
A few days for the 1h values.
Many data centers need new hardware which normally is build in? Yes, China.

mczapala, 2020-04-05:
Sure, I understand. Thank you for the information.

stevecrawshaw, 2020-04-06:
Thank you, understood. And thanks for providing this facility.

ricki-z, 2020-04-06:
At the moment the 1h stream is online again. We have to read half a million measurements for this, so the load has increased significantly.

tommasosansone91, 2020-04-18:
> Querying for the 24h data (https://data.sensor.community/static/v2/data.24h.json) returns empty json file.

Now it works, but the query on the area returns a void dictionary. E.g. See the Berlin query in the doc page https://data.sensor.community/airrohr/v1/filter/area=52.5200,13.4050,10

ricki-z, 2020-04-18:
We had to change the filter script, as someone used this in an app. This app also send massive requests with a radius of 0 ...
If using the filter it's better to use the bbox. The area filter needs to calculate the distance from the center for many sensors. This is very time consuming.

tommasosansone91, 2020-04-20:
Thanks. So which API should I use to get real-time data with the least load possible on your servers?

ricki-z, 2020-04-29:
Use the box query. And try to do a request every 2 or 3 minutes. The data returned contains all values of the last 5 minutes. So even 4 minutes may be enough to get all data.

tommasosansone91, 2020-05-03:
At the moment my app makes a request to [the real-time api address](https://data.sensor.community/static/v2/data.json) every time the user lands on a certain page (No one else than me has the app address), but I am trying to implement a scheduled script that will make a request only once every 5 minutes. I hope this latter configuration won't overload your server.

ricki-z, 2020-05-03:
This should be okay in terms of server load. The pure data.json is a "static" file, recreated once per minute.