opendata-stuttgart/meta #135: Welchen Kommunikationskanal gibt es?

mvach, 2020-04-08:
Hallo zusammen,
über die Grünen möchten wir versuchen die Anzahl der Sensoren in und um Speyer zu erhöhen und interessierte Personen darüber für Themen der Digitalisierung und OpenData gewinnen. Zudem möchten wir am danach natürlich versuchen mit den gewonnenen Daten etwas anzufangen.

Bevor wir aber Workshops planen und Sensoren zusammen stöpseln wollen wir natürlich wissen aufbauen. Daher bin ich gerade auf der Suche nach dem Kommunikationskanal von luftdaten.info.

Wenn ich die alten issues hier richtig gelesen habe gibt es aber weder mailinglist noch Forum. Wo werden denn Diskussionen über luftdaten.info technik, APIs, und weiteres geführt?

ricki-z, 2020-04-08:
Hallo,
wir sind gerade dabei, vom Namen Luftdaten.info auf Sensor.Community zu wechseln. Und auf der neuen Seite https://sensor.community/ sollten sich Kontakt-Email-Adressen finden, unter denen wir erreichbar sind. Die Seite ist aktuell noch nicht ganz fertig, weshalb die Luftdaten.info-Seite aktuell noch nicht auf die neue Seite umleitet.
Die meisten Dinge werden aktuell über Mails geklärt.
Einige der Fragen, die Sie andeuten, können hier im Wiki nachgelesen werden. Alle verwendeten, von uns geschriebenen Software-Teile sind als Sourcecode offen zugänglich. Der Rest kann bei uns nachgefragt werden. Diskussionen bezüglich der Software sind hier auf Github in den Issues des jeweiligen Repos zu finden.
Wir sind ein ehrenamtliches Projekt. Das heißt auch, das wir schon aus Zeitgründen (vom finanziellen Aufwand reden wir gar nicht erst) nicht alle Ideen umsetzen können oder es einfach etwas länger dauert. Ein Forum ist in Arbeit (und bedeutet auch viel Arbeit wenn es dann da ist), vieles läuft hier über Github, nach einer ersten Kontaktaufnahme ist auch eine Einladung zu unserem Mattermost möglich. Aber wie gesagt, bitte erst eine EMail an eine der Sensor.Community-Adressen senden.

mvach, 2020-04-10:
Hallo danke für die Antwort, ich werde meine Fragen per mail stellen.

closed 2023-02-28 by mvach