opendata-stuttgart/meta #136: API für neuste Messungen nicht verfügbar

marcelreppi, 2020-04-11:
Hallo,
die API mit der folgenden URL scheint nicht mehr erreichbar zu sein: 
http://data.sensor.community/airrohr/v1/filter/box

Gibt es eine neue API URL und eventuell eine kleine Dokumentation dafür?
Wann wird diese Funktion wieder verfügbar sein?

Etwas Hintergrundinformationen: 
Ich benutze die API aktuell um alle IDs von Sensoren zu finden, die sich in einer bestimmten bbox befinden damit ich danach alle CSV Dateien von diesen Sensoren aus dem Archiv herunterladen kann.
Gibt es vielleicht einen eleganteren Weg?

Vielen Dank für die Hilfe und dieses tolle Projekt!

ricki-z, 2020-04-13:
Hallo,

es gab einen Datenbankcrash und das Wiederherstellen hat etwas länger gedauert. Die Rechenzentren sind gerade alle etwas mehr als üblich ausgelastet.

Zum Datendownload: Wenn man nicht unnötig viele Daten herunterladen will, ist der angegebene Weg der eleganteste. Alternativ könnte man alle 3-4 Minuten die Daten über den bbox-Filter herunterladen und in eine Datenbank kopieren. In den Werten gibt es eine ID für die Messung, über die man dann Dubletten herausfiltern kann. Aber dieses Vorgehen belastet unsere Server natürlich deutlich mehr (immer hochgerechnet auf ein paar mehr Nutzer).

marcelreppi, 2020-04-13:
Danke für schnelle Antwort!

marcelreppi, 2020-04-25:
Hallo,
seit ein paar Tagen funktioniert es leider wieder nicht mehr. 
Sind die Server wieder down?

marcelreppi, 2020-07-28:
Hallo,

der Endpunkt funktioniert wieder nicht. Dieses mal kommt ein 404.

Wird der Endpunkt noch unterstützt? 
Wurde die URL geändert?

Danke und beste Grüße!

YaguraStation, 2020-07-28:
@marcelreppi Did you see #150 ? Same thing?

marcelreppi, 2020-07-28:
Yes it's the same issue.

YaguraStation, 2020-07-28:
Maybe close this rather then and follow #150

closed 2020-07-28 by marcelreppi