opendata-stuttgart/meta #137: API: filter on v2 24h average

AirK57, 2020-04-19:
Hi,

I'm looking for a way to get the AQI values via your API for one or for a group of sensors.
I think it's not possible so I'm wondering if I can filter on v2 24h average values? on a particular sensor or all sensors in a 'box'? Because it's a huge file to handle.

Thanks for your help.

ricki-z, 2020-04-20:
At the moment there is no filter for the 1h and 24h data files. For each request we have to filter exactly this huge files. And this is creating an huge work load on our servers.

tommasosansone91, 2020-05-01:
@AirK57 I am building an app in Django that is selecting sensors of interest with a custom radius-based filter. I am using the real-time data. You can find the code in this script: https://github.com/tommasosansone91/aqi_luftdaten/blob/master/pm_lookup/processing_2.py

ricki-z, 2020-05-01:
We have done this filtering nearly the same way than you did in your application (the area filter).
But look at all the arithmetic functions needed to calculate the distance and that are executed for every sensor. Its better to calculate the square around the circle and filter by lat, lon first and do all these operations on a much smaller amount of sensors after that.
This may not be much faster for single executions. But with hundreds of requests per minute you can save much compute time.