opendata-stuttgart/meta #141: Archive server very slow

marcelreppi, 2020-04-25:
Hey,

in recent days the access times and download speed for the archive has been very slow.
It did not used to be like this.
How long will this last?
 
Best regards
Marcel

ricki-z, 2020-04-26:
Hello,
we have moved the archive to another server. Can you check this server? It should show a line "User logged in, proceed." in the directory listings (DNS updates may need some time ...).

marcelreppi, 2020-04-26:
Everything working well! Thanks a lot!

Btw: The old archive URL now redirects to the new website sensor.community.
Congratulations on the new website! It looks great 👏

closed 2020-04-26 by marcelreppi

ricki-z, 2020-04-26:
We are moving all "old" server names to the new domain sensor.community step by step. Most can be easily redirected. This will be done if the new sites are working as expected (i.e. for the main site we don't have completed all translations).