opendata-stuttgart/meta #142: api v1 range lower than 1 KM not working

martinst1, 2020-05-03:
Hey guys, 

Previously that was working:
https://data.sensor.community/airrohr/v1/filter/area=42.698120,23.312889,0.3

I think it is for 300 metres.

But now everything under 1km does not work.
For example this works:
https://data.sensor.community/airrohr/v1/filter/area=42.698120,23.312889,1

Why is that and are you going to fix it?

ricki-z, 2020-05-03:
We won't "fix" this. Some people wrote "apps", that are many requests with radiuses far below 1km. What do you thing how many sensors are found with that? Zero or One, maybe two, for most areas. And for this the server needs to calulate distances between the sensors and the given center.
So people are using this sensor without thinking about the work load they create. And thats the reason why only results for radiuses larger or equal than 1km are returned.

tommasosansone91, 2020-05-16:
> Hey guys,
>
> Previously that was working:
> https://data.sensor.community/airrohr/v1/filter/area=42.698120,23.312889,0.3
>
> I think it is for 300 metres.
>
> But now everything under 1km does not work.
> For [...]

Use the map (https://sensor.community/en/) to ensure you are selecting a filter that catches something.
Furthermore, I see that Luftdaten services are experiencing big requests load, so if you can, don't use the filter api url, like the ones you showed, because they are very time consuming, and it will slow the Luftdaten server, with damages for everybody accessing it.
Instead, try to download just the whole api json (https://data.sensor.community/static/v2/data.json), and implement your own filter to select the sensors of your interest (take a look to this one whom I am still working on https://github.com/tommasosansone91/aqi_luftdaten/blob/master/pm_lookup/processing/realtime_processing.py) and maybe try to implement a scheduler that saves data into a model and avoid making a request to the server every time the users land on the page.