opendata-stuttgart/meta #143: How to manage sensor timezone in API?

tommasosansone91, 2020-05-03:
I am building an app in Django making requests to https://data.sensor.community/static/v2/data.json, parsing the content, and then exctracting data only from sensor within certain areas (I implemented a custom filter working on downloaded data).
However, from the API, I get the timestamp with 2 hours of delay with respect to my timezone (Europe/Berlin / UTC+1).
What could be the problem?

In Django settings I have:

TIME_ZONE = 'Europe/Berlin'

USE_I18N = True

USE_L10N = True

USE_TZ = True

Time settings with just my app should be fine, because when I get time from timezone.now() I get the exact time of my timezone (Europe/Berlin / UTC+1).

ricki-z, 2020-05-03:
All sensor data has timestamps in UTC. This is easier to handle as we don't have to think about summer/winter time or the timezones of sensors.

ricki-z, 2020-05-03:
And actually Berlin has the time CEST (Central European Summer Time, UTC+2).

tommasosansone91, 2020-05-14:
To whoever might need it, here is some code that helps to deal with time translation from one timezone to another and from non-summertime (default server setting) to summertime.

```
timestamp_value = sensor['timestamp']
timestamp_value = convert_datetime_timezone(timestamp_value, "Europe/London", "Europe/Berlin")                    
if time.localtime().tm_isdst != 0:
    timestamp_value = add_one_hour(timestamp_value)
```

buckfast-beekeeper, 2020-11-02:
The easiest way is to leave database data in UTC. UTC is identical over the whole world. From normal time to daylight saving in spring, there isn't a problem. There is a gap from one hour but it doesn't destroy the data. From daylight saving to normal time there is a problem. You have twice data between 2:00 and 3:00. A query with ordering on time, should be give wrong data order. Ordering on id is not good practice.  First rule for a database, the order of the records in the database make no difference for the result query. 

Maybe this problem is a non issue end of 2021 if the change is definitive history.