opendata-stuttgart/meta #144: Easily download data within location or get server connection

Nestak2, 2020-05-31:
Hi, I would like to donwnload the all-time data (~200 stations for ~2 years) for the city of Sofia, Bulgaria, and I just don't find an easy way to do this. Is there a way to filter by location or stations ids before downloading? I noticed [this issue](https://github.com/opendata-stuttgart/meta/issues/132), but it wasn't comprehensible for me - I download the mountly file for April 2020, unzipped it (13 GB) and I just can't open it, I guess it is too big.

It might be easier for me to login into the server hosting the data and download the data I need with terminal commands (I am using Linux 18.04 and know just enough about terminal commands to maybe make this work). Is the connection to the server open and can you give us some instructions on how to connect?

YaguraStation, 2020-05-31:
_Maybe_ this can help you as a start: https://github.com/YaguraStation/luftdaten-scripts

Instead of using [`get-sensor-ids.sh`](https://github.com/YaguraStation/luftdaten-scripts/blob/master/get-sensor-ids.sh) for the whole of Bulgaria, sanitise the list with only Sofia sensors and pipe to [`get-csv.sh`](https://github.com/YaguraStation/luftdaten-scripts/blob/master/get-csv.sh) with according time range. 

Gladly provide PRs with improvements. The scripts are rudimentary and only covered my specific use case, but a generalised version could benefit the whole project, I think.

EdgarMCR, 2020-06-24:
I started writing some helper functions a while back. Its in Python but feel free to copy and use if any is useful:
https://github.com/EdgarMCR/luftdaten-plotting-functions

e.g. https://github.com/EdgarMCR/luftdaten-plotting-functions/blob/master/src/get_data.py#L27