opendata-stuttgart/meta #148: 24h average turning up as empty json file

dambem, 2020-07-20:
Hi, I recently converted to using the new 24h sensor community data - https://data.sensor.community/static/v2/data.24h.json

It was working completely fine around 10 hours ago, but unfortunately it appears to be a completely empty page for the time being.

YaguraStation, 2020-07-20:
Haven't been monitoring and I'm not part of operations. But it's working for me. For you also? Or have you reached a rate limit?

dambem, 2020-07-20:
Strange, it's just showing up as empty for me. Potentially, although have asked others and they're receiving the same issue. Still can't seem to see any data when i go to the link!

dambem, 2020-07-20:
https://data.sensor.community/static/v2/data.1h.json and just data.json is working as intended which is even stranger!

ricki-z, 2020-07-20:
At the moment we have some performance issues with our database server. That's why we disabled the 24h files temporarily. This will be available again in the next days.

GerdC, 2025-11-29:
I don't know if this is old or new, https://data.sensor.community/static/v2/data.24h.json returns empty.