opendata-stuttgart/meta #149: SSL certificate error

mczapala, 2020-07-26:
Hello, when trying to use data.sensor.community v1 API, I get certificate error: SSL_ERROR_BAD_CERT_DOMAIN

tommasosansone91, 2020-07-26:
I have a similar SSLError:
HTTPSConnectionPool(host='data.sensor.community', port=443): Max retries exceeded with url: /static/v2/data.json (Caused by SSLError(CertificateError("hostname 'data.sensor.community' doesn't match 'devices.sensor.community'",),))

ricki-z, 2020-07-29:
We had to shutdown the data.sensor.community server for a while. There seems to be some sites (including our map) and tools and apps that are generating too many request in a short time.
The last days I had to split the requests over more servers. At the moment the servers are up again.

closed 2020-08-31 by ricki-z