opendata-stuttgart/meta #150: API not working?

rtg20, 2020-07-26:
Hi, this URL for v1 of the API was working great up until sometime yesterday

https://data.sensor.community/airrohr/v1/sensor/47591/

Now I get an error "The requested URL was not found on this server." - can anyone confirm? thanks.

YaguraStation, 2020-07-27:
Please specify whether it's the same as #149

rtg20, 2020-07-27:
Not to my knowledge. I haven't seen SSL-related error messages. thanks.

pixxelfriend, 2020-07-27:
Same here. I also get an 404, when i call the api endpoint for my sensor and i don´t get a cert error. I tried both API endpoints.

ricki-z, 2020-07-29:
We had to shutdown the data.sensor.community server for a while. There seems to be some sites (including our map) and tools and apps that are generating too many request in a short time.
The last days I had to split the requests over more servers. At the moment the servers are up again.

Buzzraider, 2020-08-03:
I can confirm that https://data.sensor.community/airrohr/v1/filter/ "whatever query" is not working and returns: 

"The requested URL /airrohr/v1/filter/country=DE was not found on this server.

Apache Server at data.sensor.community Port 443"

rtg20, 2020-09-13:
The URL https://data.sensor.community/airrohr/v1/sensor/47591/ is still not working, should I assume this functionality is no longer supported? What is the alternative? thanks.

ricki-z, 2020-09-14:
@rtg20 We don't receive data from this sensor. So we can't show there anything.
I.e. https://data.sensor.community/airrohr/v1/sensor/47593/ is working.

The endpoints at data.sensor.community should work again. We had some performance problems on these servers, so we had to disable the filter endpoint temporarily.