opendata-stuttgart/meta #151: ttps://data.sensor.community/airrohr/v1/filter/ 404 not found

flyinva, 2020-07-31:
Hi,

The filter URL is not reachable any more. For example https://data.sensor.community/airrohr/v1/filter/box=52.1,13.0,53.5,13.5

```
Not Found

The requested URL /airrohr/v1/filter/box=52.1,13.0,53.5,13.5 was not found on this server.
```

 Is the issue related to #150 ?

nazandrake, 2020-08-01:
I suggest implementing some kind of filter to the API for a number of connections per hour per IP to force the users to implement some kind of minimum time refresh to their apps (mine has 30min and 2km range).
Open street map API implements such kind of protection.

Buzzraider, 2020-08-03:
Hi, @flyinva looks like this issue is still present. I Can't get Your link to work, as well as mine custom filters.

Oros42, 2020-08-08:
Same thing for me :-(
Since 2020-07-26 at 3h (Paris time), https://data.sensor.community/airrohr/v1/filter/ return me a 404 for a box request.

DasNordlicht, 2020-08-10:
Is there already more information when the API will be available again? 
Or is the data retrieval now switched off?

flyinva, 2020-08-16:
The API seems online but /v1/filter/area= returns too much values !

closed 2020-11-30 by ricki-z