opendata-stuttgart/meta #152: Send data to self hosted API

umnibot, 2020-08-14:
I have tried to find information about sending the data to my own API, but cannot find information on what request to expect, so I can build my logic.

Is there any documentation as of what the data,php should look like? Or at least is the request GET or POST

[image]

Thanks

umnibot, 2020-08-15:
Okay, I found out that it is a simple POST. However, when I try to just send it to my server, it says Connection refused.
But I can access the data.php from any other server.

Do is this option Send data to custom API even works?

ricki-z, 2020-08-29:
This option is working. Is the sensor in the same network as this server or in a guest wifi? 
You also need to activate the option "Send data to custom API"(but I think you did this and only the screenshot was made with the option disabled)

umnibot, 2020-08-31:
Hello, the sensor is within my home network, while the API is on my webserver.
Yes, I did check the option "Send data to custom API".
I have tried all possible scenarios that I could think of (HTTP, https, different ports, different pats, combined paths). Also, install a logger script where the API is, to see if there is a connection attempt.

if I execute the URL via the browser it logs the activity, but no attempts if I try it via the sensor.

umnibot, 2020-08-31:
This is what I have in device status

[image]

ricki-z, 2020-08-31:
So your webserver is in your local network (192.168.234.1)?
Can you see any access in your webserver logs?

umnibot, 2020-08-31:
No, my server is not in my local network. It is an outside network

ricki-z, 2020-08-31:
Then please enter the correct IP of the server (192.168.234.1 is your router I think) or the hostname.

umnibot, 2020-08-31:
Well, I don't know where you are getting this IP address.
in the server field, I have the hostname of my server
[image]

umnibot, 2020-08-31:
Oh, you got it from the initial screenshot. That's the configuration before me editing it.

ricki-z, 2020-09-01:
remove the http:// at the start f the line. the protocol is choosen by port. There must be only the server name or IP in this field.

umnibot, 2020-09-01:
Same result with a configuration like this.

[image]

ricki-z, 2020-09-02:
Really?
Server: Only the server name or server IP !!!!
Path: the complete path to the custom API script, in your case /App/AirQData/data.php

umnibot, 2020-09-13:
is there anything else I can try. Still don't receive any data from the sensor to my server

ricki-z, 2020-09-14:
@umnibot can you send me the hostname and script path to rajko (at) sensor.community? So I can have a look at that.

ricki-z, 2020-09-14:
The server is forwarding to https. Could you please activate https for this API and check again? It's possible that the firmware isn't following the 30x redirect but also doesn't show an error (http error codes are greater or equal 400).

umnibot, 2020-09-14:
Just tried with https and got the connection refused
--

ricki-z, 2020-09-14:
Our firmware only supports the following ciphers (minimum set of ciphers):
TLS_RSA_WITH_AES_128_CBC_SHA256,
TLS_RSA_WITH_AES_256_CBC_SHA256,
TLS_RSA_WITH_AES_128_CBC_SHA,
TLS_RSA_WITH_AES_256_CBC_SHA
We can't enable all possible ciphers as this would need too much flash and RAM.

umnibot, 2020-09-14:
So, the only option, if I cant support these ciphers, would be to make it HTTP:. Right?

umnibot, 2020-09-14:
Just activated TLS_RSA_WITH_AES_128_CBC_SHA256 but still can't make it work

ricki-z, 2020-09-14:
At the moment: yes.
We have to check the memory consumption of additional ciphers. My last info was that we need to activate all cipher if we need others. That would really need too much flash.
Another factor is the time for de- and encoding. Other ciphers are so complex that the NodeMCU needs too long.

umnibot, 2020-09-15:
Okay, I will set the folder to http to see if it will work.

umnibot, 2020-09-15:
Ok, I have managed to add make it working. I have a question about a few data keys marked in red. What is that measurement?

[image]

Also in the device status, I see read timeout. Is that normal?

[image]

ricki-z, 2020-09-15:
The measurements "samples", "min_micro" and "max_micro" are used to check the overall performance of the firmware. "samples" is the count of loops the main() function was running, "min_micro" and "max_micro" are the minimum and the maximum time the main() loop was running (in microseconds).

To the screenshot: Only the last errors are shown, not the result of the lastest transmission. So the "read timeout" was only a one time failure. See the difference between "number of measurements" and "data send".

ricki-z, 2020-09-15:
Regarding the problem with HTTPS: Please check your server at https://www.ssllabs.com/ssltest/analyze.html . As you can see the ciphers needed aren't supported yet.

manne01, 2021-07-25:
Hallo, es hat mich grade leider einige Zeit gekostet rauszufinden, was mit meinem Feinstaubsensor los war. Auch bei mir lags am "zu modernen SSL Profil" am empfangenden Webserver meiner Custom API.
Leider hat das Logging im Feinstaub sensor garnicht geholfen. Es stand (max Info) immer nur dran: Sending Custom API (nach den öffentlichen APIs natürlich;-) )
## Sending to custom API
time for sending: 6ms

Im "Fehler" modus des log dasselbe.

Es wäre klasse, wenn das Logging für solche Verbindungsprobleme zB den Return Code oder sowas ins Log schreiben würde.

Bei mir Firmware NRZ-2020-133/DE vom Nov 29 2020 hab ich es nicht geschafft die Verbindung zur Custom API damit zu debuggen.

Erst diese Diskussion hier brachte mich auf den richtigen Weg. Besten Dank dafür!

umnibot, 2021-09-12:
Hi Rajko. It seems like with firmware update NRZ-2020-133 the sensor stopped sending info to the custom server.  Now looking at my server log it seems that the last was data sent on 2020-12-14 16:13:43

[image]

Nothing on the server was changed nor on the sensor (using http not https).
In the sensor debug dont see anything to tell me what is going on.

ws: root ...
ws: status ...
ws: values ...
PM10:  13.93
PM2.5: 8.32
----
## Sending to sensor.community - SDS011
Succeeded - api.sensor.community
Temperature (°C): 27.02
Pressure (hPa): 94872.22
Humidity (%): 23.94
----
## Sending to sensor.community - BME280
Succeeded - api.sensor.community
## Sending to madavi.de: 
Succeeded - api-rrd.madavi.de
## Sending to custom api: 
Time for Sending (ms): 5873
PM10:  8.20
PM2.5: 5.55
----

The sensor is not even trying to access the custom API. If I try to access the API over a browser, there are no restrictions and I see the attempt in the server log (but not see attempt from the sensor).
Was there anything changed or sould I change somethng?

Thanksd

ricki-z, 2021-09-12:
Please check that the SSL certificate is max. 2048 bit. And that the supported ciphers contain one of the following:
TLS_RSA_WITH_AES_128_CBC_SHA256,
TLS_RSA_WITH_AES_256_CBC_SHA256,
TLS_RSA_WITH_AES_128_CBC_SHA,
TLS_RSA_WITH_AES_256_CBC_SHA

umnibot, 2021-09-12:
Do I need SSL for HTTP only ?

umnibot, 2021-09-12:
Also just checked that we have TLS_RSA_WITH_AES_128_CBC_SHA256

umnibot, 2021-09-12:
[image]

ricki-z, 2021-09-12:
SSL isn't needed. But in most cases the SSL cert and the missing ciphers are  the cause for the not working self hosted API.
You could check if it's working without SSL. 
Do you have access to the error logs of the web server? Maybe there is a hint why the data isn't saved.

umnibot, 2021-09-13:
It doesnt working with or without SSL. and in the server logs there is no evidence of an attempt for trying to connect.
How does an attempt for sending data from the sensor should look like?

pjgueno, 2021-09-13:
Check the end points and the sensor number X-Pin and X-sensor well.

[https://github.com/opendata-stuttgart/meta/wiki/EN-APIs](https://github.com/opendata-stuttgart/meta/wiki/EN-APIs)

It should work with a normal POST request.

**SC server:**

`{
  "software_version": "<WHAT YOU WANT>", 
   "sensordatavalues":[
    {"value_type":"P1","value":"<PM10>"},
    {"value_type":"P2","value":"<PM2.5>"}
  ]
} `

`{
  "software_version": "<WHAT YOU WANT>", 
   "sensordatavalues":[
     {"value_type":"temperature","value":"<TEMP>"},
     {"value_type":"pressure","value":"<PRESS>"},
     {"value_type":"humidity","value":"<HUMI>"}
  ]
} `

**Madavi server:**

`{
  "software_version": "<WHAT YOU WANT>", 
   "sensordatavalues":[
    {"value_type":"SDS_P1","value":"<PM10>"},
    {"value_type":"SDS_P2","value":"<PM2.5>"}
  ]
} `

`{
  "software_version": "<WHAT YOU WANT>", 
   "sensordatavalues":[
     {"value_type":"BME280_temperature","value":"<TEMP>"},
     {"value_type":"BME280_pressure","value":"<PRESS>"},
     {"value_type":"BME280_humidity","value":"<HUMI>"}
  ]
} `

umnibot, 2021-09-13:
The problem is sending the data to a custom endpoint that was working before and not working now. So no idea how Madavi and SC server values can help me here. I already know what to expect and can see my values https://data.sensor.community/airrohr/v1/sensor/40205/

ricki-z, 2021-09-14:
There wasn't any change to the release firmware in the last months. So this shouldn't be the cause of the problem.
Maybe there were any updates for PHP on your hosting platform. Or there is any "firewall" or "antispam" tool on the way from the sensor to the server that is blocking the request.

umnibot, 2021-10-04:
Hello Rajko,

It seems that my sensor even stopped sending data to https://data.sensor.community/airrohr/v1/sensor/40205/ 
The dataset comes empty

**Here is the log**
Temperature (°C): 18.33
Pressure (hPa): 95571.09
Humidity (%): 39.80
----
## Sending to sensor.community - Bws: root ...
ws: values ...
ws: root ...
PM10:  10.60
PM2.5: 5.30
----
## Sending to sensor.community - SDS011
Temperature (°C): 16.50
Pressure (hPa): 95650.38
Humidity (%): 42.27
----
## Sending to sensor.community - BME280
## Sending to madavi.de: 
## Sending to custom api: 
Time for Sending (ms): 100922