opendata-stuttgart/meta #156: Dokumentation für Filter

JavaNoob-dotcom, 2021-01-22:
Hi, 

gibt es eine Dokumentation der API in der sämtliche Filter und weitere Parameter beschrieben werden? 
Ich habe bisher immer nur zufällig in verschiedenen Posts Endpoints gefunden wie z.B. data.sensor.community/airrohr/v1/filter/country=DE&type=Laerm.
Die Dokumentation unter https://github.com/opendata-stuttgart/meta/wiki/APIs bietet ja leider keine Informationen über den Abruf von Daten, sondern nur wie man Daten publishen kann.

Ich hoffe jemand kann mir weiterhelfen, vielen Dank schonmal.

pjgueno, 2021-01-22:
https://github.com/opendata-stuttgart/meta/wiki/EN-APIs

JavaNoob-dotcom, 2021-01-22:
Vielen Dank, hatte das wohl immer wieder überlesen.

closed 2021-01-22 by JavaNoob-dotcom