opendata-stuttgart/meta #158: what happened to type=Laerm ?

derkod, 2021-04-16:
I use this URL to get all the Noise-sensors in NL: https://data.sensor.community/airrohr/v1/filter/country=NL&type=Laerm
But since a few days it does not work anymore.
Does anyone know what happened and how I should have known that?
Thanks!

ricki-z, 2021-04-16:
Sorry we needed to rename this type. In all instructions it's refered to as DNMS, so the type is now 'DNMS (Laerm)'.

derkod, 2021-04-16:
OK. I tried this from Python:
url='https://data.sensor.community/airrohr/v1/filter/country=NL&type=%27DNMS%20(Laerm)%27'
and
url="https://data.sensor.community/airrohr/v1/filter/country=NL&type='DNMS (Laerm)"
and many other variations with single quotes/double quotes/spaces/non spaces
Also tried it from a webbrowser.
But still no data. Even if I change NL in DE
What do I miss here?

ricki-z, 2021-04-21:
This should work again.
The correct URL for the request would be:
https://data.sensor.community/airrohr/v1/filter/country=NL&type=DNMS%20(Laerm)
The quotation marks were only written to show start and end of the query string.

derkod, 2021-04-21:
Yes, it works now. Thanks for resolving this!

closed 2021-04-21 by pjgueno