opendata-stuttgart/meta #164: Local HTTP-Request to Device possible? How to get a JSON-String back?

DeepHyperspace, 2022-04-30:
I have recently put my fine dust sensor into operation. I wanted to know if it is possible to send a local HTTP request to the device itself, instead of using the webapi, so that you are independent of an internet connection.

I am a big fan of Node-Red, the following tries show what I tried. Using the WEB-API I get a nice JSON object back of course. (Screenshot 1)
I tried it with the http://192.168.178.44/values, but there I get the whole HTML code back, I could of course read the values out of it, but maybe there is a URL that returns a JSON string that I could interpret easier. Does this feature exist?

[image]
[image]

ricki-z, 2022-05-01:
To read a JSON directly from the device you need to use the address http://192.168.178.44/data.json
There is also an endpoint for prometheus metrics which is also using JSON format.

DeepHyperspace, 2022-05-01:
Thanks a lot that Was exactly what I was searching for 😊
I was not able to find that Informationen, is that documented somewhere?

closed 2022-05-01 by DeepHyperspace