opendata-stuttgart/meta #165: The requested URL /airrohr/v1/sensor/45802 was not found on this server.

NeetaRJ, 2022-05-06:
Hi,

The API URL is not reachable any more. For example https://data.sensor.community/airrohr/v1/sensor/5206363

[image]

pjgueno, 2022-05-06:
The sensor ID is too long. Are you sure of it? Check on the map. And put a slash at the end of the url.

NeetaRJ, 2022-05-06:
[image]
This is my sensor on MAP.
I am confused about the api id.
[image]

What needs to be passed as a value for api id https://data.sensor.community/airrohr/v1/sensor/{apiID}/

pjgueno, 2022-05-06:
You have to use 72267.
It works. I checked on my computer.