opendata-stuttgart/meta #168: 4G/5G without WIFI

bf, 2023-07-12:
Hi, 

love the project. Has someone tried to run the sensor with a steady power source but without WIFI?
Maybe someone has done it and you can point me in the right direction.
I've searched the github repo and google for "4g", "5g", "airrohr 4g" and "airrohr 5g" but did not find anything.

Thank you!

bf, 2023-07-12:
I just noticed that airrohr supports lorawan on arduino feather m0, this answers my question: https://github.com/sensebox/airrohr-sensors-software/blob/master/esp8266-arduino/ppd42ns-wificonfig-ppd-sds-dht/Readme.md

closed 2023-07-12 by bf

pjgueno, 2023-07-12:
Please check:
https://github.com/aircarto/NebuleAir_NBIOT

It was fully tested.

Feel free to contact me for further info.