opendata-stuttgart/meta #170: Sensor Daten not visible on map

skycryer, 2025-01-18:
I did setup my devices today, getting data on the ui, registered with esp8266-8658704 but no data is send. Seeing connection failed error for data send/return. 

Any idea?

ricki-z, 2025-02-02:
Hello,
could you please check that your firmware version is from 2024. Older versions can't send via HTTPS as there was a change in the certificate chain.
As a temporary solution you could deactivate https for the APIs.

(For any further questions please use our forum at https://forum.sensor.community/)

skycryer, 2025-02-02:
I just installed it from your website in the day I wrote the first Mail.

Cheers

ricki-z, 2025-02-02:
I can see your sensor here: https://maps.sensor.community/#15/50.6680/6.4780

skycryer, 2025-02-02:
Yes, I already got email support and in the end I had to add manual IP
configuration then it worked. Forgot to close the ticket then.

closed 2025-02-23 by ricki-z